yuanjun-gao/ICCV2023_SLAM_Challenge #3: timestamp of ground truth

zqlee-ronghui, 2023-08-19:
Hi, I was curious about the timestamp of the ground truth as I received some errors despite outputting trajectory with the timestamp from the IMU Msg's header stamp.

Let's say,

- SubT_MRS_Final_Challenge_UGV3.txt does not meet the duration requirement.
- TartanAir_lidar_* found no matching timestamps with time offset about 90569539.49203s

water-horse, 2023-08-20:
Hi. Sorry for any inconvenience caused. For the SubT_MRS_Final_Challenge_UGV3, we've observed an issue with the ROS bags. We've made updates regarding this issue on [this website](https://superodometry.com/iccv23_challenge_LiI). See 'Important Latest Updates' close to the beginning. For The second problem regarding the scoring of TartanAir_lidar_*, we’ve fixed the time stamp bug. There should be no such error now after submission.

zqlee-ronghui, 2023-08-21:
ok, thanks

closed 2023-08-21 by zqlee-ronghui