yuanjun-gao/ICCV2023_SLAM_Challenge #4: unable to register the gradescopt account

samsdolphin, 2023-08-22:
[image]
Hi organizers, I cannot register the account at the [Gradescope](https://www.gradescope.com/), which says the invalid course code.

xiangfuli, 2023-08-22:
Hello samsdolphin, thanks for the feedback. I've tried to used the course Entry Code to enroll in the gradscope and everything went smoothly. Could you try again? 

Or if you can provide your email address, I can manually add you into the course. Thanks.

Jerry-jing, 2023-08-24:
> 您好，samsdolphin，感谢您的反馈。我尝试使用课程入门代码注册 gradscope，一切都很顺利。你能再试一次吗？
> 
> 或者，如果您可以提供您的电子邮件地址，我可以手动将您添加到课程中。谢谢。
[image]

您好，我也遇到了一样的问题，我的邮箱是izhang_jing@163.com，您可以帮我添加到课程中吗，谢谢您

water-horse, 2023-08-24:
> 您好，我也遇到了一样的问题，我的邮箱是[izhang_jing@163.com](mailto:izhang_jing@163.com)，您可以帮我添加到课程中吗，谢谢您

izhang_jing@163.com is added to the visual inertial track

wwtinwhu, 2023-08-25:
I encounted the same issue. I fixed it by using Carnegie Mellon University as school. @samsdolphin @water-horse @xiangfuli

samsdolphin, 2023-08-25:
> Hello samsdolphin, thanks for the feedback. I've tried to used the course Entry Code to enroll in the gradscope and everything went smoothly. Could you try again?
> 
> Or if you can provide your email address, I can manually add you into the course. Thanks.

Yes, thanks for your feedback, so this is my email xliuaa@connect.hku.hk.

xiangfuli, 2023-08-25:
> > Hello samsdolphin, thanks for the feedback. I've tried to used the course Entry Code to enroll in the gradscope and everything went smoothly. Could you try again?
> > Or if you can provide your email address, I can manually add you into the course. Thanks.
> 
> Yes, thanks for your feedback, so this is my email [xliuaa@connect.hku.hk](mailto:xliuaa@connect.hku.hk).

I've added you into the course.

xuwuzhou, 2023-08-31:
How can I do with issue https://github.com/water-horse/ICCV2023_SLAM_Challenge/issues/6
@xiangfuli

closed 2023-09-07 by samsdolphin

xiangfuli, 2023-09-25:
> tee.shutong@gmail.com

Already added you into the courses.