yuanjun-gao/ICCV2023_SLAM_Challenge #7: Test Failed

letsgofeng, 2023-08-30:
Hi, after submitting the trajectory results, the following error is printed, what is the reason for this?

ERROR] csv file /autograder/submission/visual_inertial_track/SubT_MRS_Laurel_Caverns_Handheld1.txt does not exist
Failed to check the time duration of trajectory /autograder/submission/visual_inertial_track/SubT_MRS_Laurel_Caverns_Handheld1.txt. Please check your file format or path.
--------
[ERROR] csv file /autograder/submission/visual_inertial_track/SubT_MRS_Laurel_Caverns_Handheld2.txt does not exist
Failed to check the time duration of trajectory /autograder/submission/visual_inertial_track/SubT_MRS_Laurel_Caverns_Handheld2.txt. Please check your file format or path.
--------
[ERROR] csv file /autograder/submission/visual_inertial_track/SubT_MRS_OverExposure_LegRobot.txt does not exist
Failed to check the time duration of trajectory /autograder/submission/visual_inertial_track/SubT_MRS_OverExposure_LegRobot.txt. Please check your file format or path.
--------
[ERROR] csv file /autograder/submission/visual_inertial_track/TartanAir_visual_endofworld.txt does not exist
Failed to check the time duration of trajectory /autograder/submission/visual_inertial_track/TartanAir_visual_endofworld.txt. Please check your file format or path.
--------
[ERROR] csv file /autograder/submission/visual_inertial_track/TartanAir_visual_moon.txt does not exist
Failed to check the time duration of trajectory /autograder/submission/visual_inertial_track/TartanAir_visual_moon.txt. Please check your file format or path.
--------
[ERROR] csv file /autograder/submission/visual_inertial_track/TartanAir_visual_westerndesert.txt does not exist
Failed to check the time duration of trajectory /autograder/submission/visual_inertial_track/TartanAir_visual_westerndesert.txt. Please check your file format or path.
--------

Test Failed: True != False : Error happened. Please refer to the posted information.

xiangfuli, 2023-08-30:
Please compress the folder that contains the whole trajectory result files as the zip file. 

You can refer to the packaging details in the [super odometer website](https://superodometry.com/iccv23_challenge_VI).

HLkyss, 2023-09-08:
meet the same problem. I compress all of the txt files into one "visual_inertial_track.zip" directly and this error keeps coming up. Have u solved it? Besides, I also tried to use csv instead of txt and still can't solve it.

letsgofeng, 2023-09-09:
I used the following directory compression to successfully submit it.
"visual_inertial_track\visual_inertial_track\*"
This should be related to the official decompression command.

HLkyss, 2023-09-09:
Thank you so much for replying to my question, but i still fail to submit it. After i drop my compressed file to Gradescope, the information on the "Submit Programming Assignment" page is like "visual_inertial_track/visual_inertial_track/SubT_MRS_Laurel_Caverns_Handheld1.txt", "visual_inertial_track/visual_inertial_track/SubT_MRS_OverExposure_LegRobot.txt" and so on, according to your suggestion, and then get the same error. Maybe i misunderstand your suggestion, or maybe there's something else wrong with my file? Anyway, thanks again for your reply!
[image]

letsgofeng, 2023-09-09:
Judging from your submission information, you don't need to compress the file as I suggested.
After uploading the compressed package, the correct path is shown as: visual_inertial_track/SubT_MRS_Laurel_Caverns_Handheld1.txt

xiangfuli, 2023-09-11:
Please compress the folder that contains the whole trajectory result files as the zip file. For example, create a directory named `visual_inertial_track` and put all your trajectory results inside. Them compress the directory using command `zip visual_inertial_track.zip visual_inertial_track`.

You can refer to the packaging details in the [super odometer website](https://superodometry.com/iccv23_challenge_VI).

YibinWu, 2023-09-20:
I had the same issue when I tried to upload the lidar_inertial_track. I have already prepared the .zip file according to the instructions.

xiangfuli, 2023-09-20:
> I had the same issue when I tried to upload the lidar_inertial_track. I have already prepared the .zip file according to the instructions.

Your trajectory result file was larger than usual so the grader killed itself caused by OOM. I'm upgrading the grader. You  may check you test result later.

YibinWu, 2023-09-21:
> > I had the same issue when I tried to upload the lidar_inertial_track. I have already prepared the .zip file according to the instructions.
> 
> Your trajectory result file was larger than usual so the grader killed itself caused by OOM. I'm upgrading the grader. You may check you test result later.

Hi, now it works. But there is an issue about the time duration. I got this 
"_Trajectory /autograder/submission/lidar_inertial_track/SubT_MRS_Urban_Challenge_UGV2.txt does not meet the duration requirement. Submitted trajecotry's duration is 15 seconds shorter/longer than ground truth trajectory's duration_."
But I checked that sequence, the rosbag start time and end time are 1582405190.89 and 1582408367.82, while my trajectory starts from 1.582405190065000022e+09 and end at 1.582408366699999958e+09, which are quite close (less than 15s). Could you help me to figure it out?

YibinWu, 2023-09-21:
> > > > I had the same issue when I tried to upload the lidar_inertial_track. I have already prepared the .zip file according to the instructions.
> > > 
> > > 
> > > Your trajectory result file was larger than usual so the grader killed itself caused by OOM. I'm upgrading the grader. You may check you test result later.
> > 
> > 
> > Hi, now it works. But there is an issue about the time duration. I got this "_Trajectory /autograder/submission/lidar_inertial_track/SubT_MRS_Urban_Challenge_UGV2.txt does not meet the duration requirement. Submitted trajecotry's duration is 15 seconds shorter/longer than ground truth trajectory's duration_." But I checked that sequence, the rosbag start time and end time are 1582405190.89 and 1582408367.82, while my trajectory starts from 1.582405190065000022e+09 and end at 1.582408366699999958e+09, which are quite close (less than 15s). Could you help me to figure it out?
> 
> The duration should be around 3175.9292969703674s, but yours trajectory ends at 1582406649.43s and overall duration is 1459.3650000095367s evaluated by command 'evo_traj'. You can download the `evo` tools and check you result by yourself.

Hi thanks for your reply. I checked my results. It has around 3176s. 
[image]

YibinWu, 2023-09-21:
> Hi thanks for your reply. I checked my results. It has around 3176s. [image]

Hi I have updated the results. Now it is fine. Thanks.

xiangfuli, 2023-09-21:
> Hi I have updated the results. Now it is fine. Thanks.

UR welcome.

closed 2023-09-27 by water-horse