yuanjun-gao/ICCV2023_SLAM_Challenge #13: evaluative method

zhangxiao19980516, 2023-09-13:
Hello，I have two questions。first，can you give the code of the evaluation？if not，can you give the formula and units of evaluation。second，Some slam algorithms need to be initialized in motion(vins-mono), in which case, how should the initial position be considered?

shibowing, 2023-09-16:
Hi Zhangxiao, 

Sorry for late reply because of ICRA deadline. We will post the formula and units soon. 

For question2, we will manually align your estimated trajectory with our ground truth trajectry. 

Feel free to let me know if you still have questions. ^^

zhangxiao19980516, 2023-09-17:
For the second problem, there is no sensor pose in the initialization part, so there will be a part of the trajectory missing in the initial part, so what to do with this small segment of the trajectory? Will the pose of the missing trajectory be filled in with 0 or with the first estimated pose?

water-horse, 2023-09-18:
You are right. The last row of the thermal camera's calibration file should be 0, 0, 0, 1

shibowing, 2023-09-18:
Hi @zhangxiao19980516,  the pose of the missing trajectory be filled in with 0. 

> For the second problem, there is no sensor pose in the initialization part, so there will be a part of the trajectory missing in the initial part, so what to do with this small segment of the trajectory? Will the pose of the missing trajectory be filled in with 0 or with the first estimated pose?

Hi @zhangxiao19980516 , Thanks for reporting issues and will fix it now. For previous question, The missing trajectory will be filled with 0.

zhangxiao19980516, 2023-09-19:
Thank you very much for your answer. What the means of  four IMU simulation noise parameters provided in question #2 ？
[image]

Amigoshan, 2023-09-22:
We add bias using a random walk algorithm. The above parameters are used in the random walk model.

MukilSaravanan, 2023-09-23:
Following my own question,  https://github.com/water-horse/ICCV2023_SLAM_Challenge/issues/2#issuecomment-1731840943 
Could you please where to find IMU parameters (such as gyro, accelero noise density, random walk)?

shibowing, 2023-09-24:
Hi @MukilSaravanan ,
Sorry for late reply.
The SubT-MRS dataset can provide a reference IMU parameters. However, we cannot provide the calibrated one since the datasets are collected from different time and different robots. For TantarAir, we should have IMU parameters and will upload later. Hi @Amigoshan, could you please post the IMU parameters again?

MukilSaravanan, 2023-09-25:
> Hi @MukilSaravanan , Sorry for late reply. The SubT-MRS dataset can provide a reference IMU parameters. However, we cannot provide the calibrated one since the datasets are collected from different [...]

Could you please post it here once? Thanks

water-horse, 2023-09-25:
@MukilSaravanan 

> Thank you very much for your answer. What the means of four IMU simulation noise parameters provided in question #2 ？ [image]

> We add bias using a random walk algorithm. The above parameters are used in the random walk model.

closed 2023-12-16 by yuanjun-gao