yuanjun-gao/ICCV2023_SLAM_Challenge #17: Evaluation & Dataset question

Natsu-Akatsuki, 2023-09-19:
@water-horse, @shibowing
Hello! 
1. As outlined in the 'Submit the Results' section on the official website (Sensor Fusion SLAM Challenge), does the statement 'The number of lines/poses must match the number of image frames in that trajectory' imply that the timestamp of the ground truth pose should coincide with the timestamp of the image? Therefore, would it be more appropriate to utilize the image timestamp rather than the IMU timestamp or the LiDAR timestamp?
2. In both the flashlight dataset and the long_corridor dataset, we have observed a phenomenon as illustrated below (using the flashlight dataset as an example). Do we need to predict poses for this missing-data time window? Should the description of 'Duration' in the website's table be updated to reflect this?"
[image]

water-horse, 2023-09-20:
Answer to question 1: interpolation on translations will be performed if the timestamps are not strictly the same. Typically the errors incurred by interpolation does not matter because we will apply different thresholds when converting trajectory errors to scores

Answer to question 2: you can ignore the beginning and ending tf messages whose timestamps are not covered by IMU, image, and LiDAR messages.

Natsu-Akatsuki, 2023-09-20:
ok, thx

closed 2023-09-20 by Natsu-Akatsuki