yuanjun-gao/ICCV2023_SLAM_Challenge #19: Report

YibinWu, 2023-09-22:
Hi there,
**Will the report be open to the public somewhere at some point or it will be kept confidential?** 
**Is it possible to submit it later?** 
Our method is involved in an unpublished paper that is under review. We don't want it to be considered a dual submission. 
Thanks a lot.

shibowing, 2023-09-24:
Hi Yibin,

Good to hear from you. 

Unfortunately, the report cannot be submitted later since we need to be fair for all the teams.

However, the report can be confidential now. If we want to public somewhere in future, we will contact you first and ask your suggestions. 

Thank you! 

Best,

letsgofeng, 2023-09-24:
Hi, when is the deadline for the report submission?

shibowing, 2023-09-24:
Hi @letsgofeng , the report deadline is 09/25/2023.

closed 2023-12-16 by yuanjun-gao