yuanjun-gao/ICCV2023_SLAM_Challenge #22: Deadline Extended

water-horse, 2023-09-26:
Hello everyone!

Thanks for participating SLAM challenge !

As the competition deadline approaches, many teams asked for an extension. To ensure fairness, we've extended **the deadline to 11:59 PM EDT, October 15th, 2023**. It is important to note that each team **must provide answers to all questions in the report template**. A high-quality report will also be **factored into your final score** and assessed as a reference for the submitted trajectory.

Starting from September 25th, each team is limited to a maximum of 30 submissions. Please be aware that exceeding this limit will result in the **30th submission being considered as the final one**.

Additionally, registering multiple accounts is strictly prohibited now. Feel free to let us know if you have more questions.

We wish you the best of luck in the challenge! ^^

Natsu-Akatsuki, 2023-09-26:
Hello, we have a couple of questions:

**[Leaderboard]** 1. Any information about when the leaderboard will become available?

**[Evaluation]** 2. Can you clarify how the evaluation process manages predicted poses that occur before the starting timestamp of the ground truth poses? We've observed that it doesn't appear to exclude these predicted poses, when we introduce abnormal values or zero poses (as mentioned in [#issue13](https://github.com/water-horse/ICCV2023_SLAM_Challenge/issues/13)) prior to the initial ground truth pose. This appears to have an impact on the APE and RPE scores.

water-horse, 2023-09-27:
1. The leaderboard will be available on Sep 28.
2. The beginning of the predicted poses and the ground truth poses are automatically aligned. The ground truth poses start when all the sensor data become available.
Will post here if there are some updates about the evaluation process.

closed 2024-10-29 by shibowing