yuanjun-gao/ICCV2023_SLAM_Challenge #24: General Feedback

davidskdds, 2023-10-09:
Hello,
I am PhD student in the SLAM area and wanted to participate in your SLAM challenge.
After attending in the Hilte SLAM challenges 2022 and 2023 with my pipeline I also took a look at your datasets.
Unfortunately, I noticed a few difficulties that, in my case, reduced the motivation to participate:

1. the sequences from the TartanAir dataset lack timestamps per point. Only in another Github issue it was communicated that the scans were generated statically in each case. However, the static generation of the scans does not fit the reality and prevents the easy application of common LiDAR inertial SLAM approaches.

2. the sequences from the TartanAir dataset are multiple times provided with a different noise identifier. Here it is unclear which Rosbags are actually relevant.

3. only velodyne scans are included in the Rosbags from the SubT-MRS dataset. This is a manufacturer-specific format. Here it would be easier if the data had already been converted into PointCloud2 messages in advance, as this format is more common.

All in all, due to the points mentioned above, participation entails some additional development effort, which in my case reduces the willingness to participate.

I don't want to criticize you, but to share my perception in a productive way, in order to possibly reach more participants in future SLAM Challenges.

Kind regards 
David Skuddis

shibowing, 2023-10-11:
Hi @davidkdds, 

Thank you so much for the feedback! We appreciate your kind advice. 

We will provide the solutions in the next 36 hours. 

Best, 

Shibo

water-horse, 2023-10-11:
Instructions for converting velodyne_msgs/VelodyneScan to sensor_msgs/PointCloud2 are updated on [README.md](https://github.com/water-horse/ICCV2023_SLAM_Challenge/blob/master/README.md)

shibowing, 2023-10-14:
Hi @davidskdds ,

Thanks for your feedback. We did some quick fix for your questions and hope it can improve the participation experience.

1.  I have reached out to Wenshan regarding the Tartan Air datasets and the timestamp issue. As the data is collected in simulation environments, each data point is timestamped simultaneously. However, I acknowledge your point, and we will address this concern by providing new datasets that aim to better simulate real-world lidar scans.

2.  That confused datasets are removed in the google drive. Thanks for feedback 

3. Since our datasets containts multi-modalities, the PointCloud2 messages will make rosbag very large and takes a lot time to download. Therefore, we added intructions to convert the velodyne_msgs/VelodyneScan to sensor_msgs/PointCloud2. This is just simply **launch a rosnode** to subscribe the VelodyneScan and publish PointCloud2 message. I think it is easy to install and hope it can make your life easier. 

Instructions for converting velodyne_msgs/VelodyneScan to sensor_msgs/PointCloud2 are updated on [README.md](https://github.com/water-horse/ICCV2023_SLAM_Challenge/blob/master/README.md)

Thanks again for your suggestions. Feel free to let me know if you have other suggestions! The datasets is still improving and maintaining for long-term!

closed 2023-12-16 by yuanjun-gao