yuanjun-gao/ICCV2023_SLAM_Challenge #26: certificate

haoxinnihao, 2023-10-20:
After the seminar, will grade proof of competition results be provided?such  as certificate.

shibowing, 2023-10-23:
Yes, we are making the certificate and it may take 2 weeks to finish it. We will send it through email.

zhangxiao19980516, 2023-10-24:
Hello, is the name on the certificate the team name? Or the name of the member on the report.

shibowing, 2023-10-25:
We will show the name of member. ^^

samsdolphin, 2023-11-10:
> Yes, we are making the certificate and it may take 2 weeks to finish it. We will send it through email.

Hi @shibowing, any updates on this?

zhangxiao19980516, 2023-11-15:
Hello @shibowing  When will the certificate be issued?

shibowing, 2023-11-15:
Hi all, sorry for the delay. I just received  the email from the CMU professor. The certificate will be sent on Nov 18th.

shibowing, 2023-11-15:
Hi all,

Sorry for the delay.  We are nearly there; just one more professor's signature is needed. I received confirmation a few hours ago that this will be completed within the week.

As such, I anticipate that we will be able to send out the certificate by November 18th. On the same day, I'll also provide instructions for you on the winner team prize.

Thank you very much for your patience and understanding during this process.

Best regards,

closed 2023-12-16 by yuanjun-gao