yuanjun-gao/ICCV2023_SLAM_Challenge #27: Leaderboard update?

samsdolphin, 2023-10-26:
Dear Organizers of the ICCV SLAM Challenge,

Thank you for your effort in organizing such a great SLAM competition.

When will the **official website** of this challenge and the **leaderboard** be updated since the team name is currently my GitHub account name?

Thank you!

shibowing, 2023-10-29:
Hi @samsdolphin, 

Thanks for reminder! We will update the website recently.

samsdolphin, 2023-11-10:
> Hi @samsdolphin,
> 
> Thanks for reminder! We will update the website recently.

Hi @shibowing, any updates on this?

shibowing, 2023-11-12:
Hi @samsdolphin , Thanks for asking. We are getting signature from CMU professors on your certificates. There is only one profressor left and I just sent email to him for progress. Once it is done, I will directly send it through email.

yuanjun-gao, 2023-12-16:
Hi @samsdolphin, the LIO and VIO leaderboard is updated.

closed 2023-12-16 by yuanjun-gao