yuanjun-gao/ICCV2023_SLAM_Challenge #29: Where is the SubT-MRS dataset ?

diamonazreal, 2023-11-23:
Where can I find the SubT-MRS multi-robot dataset？

shibowing, 2024-02-06:
Hi @diamonazreal, 

The focus of this challenge does not focus on multi-robot systems. Nevertheless, we released small collection of multi-robot data. Specifically, the datasets titled "Final_Challenge_UGV1," "Final_Challenge_UGV2," and "Final_Challenge_UGV3" are multi-robot scenarios. We anticipate releasing additional datasets in the forthcoming months. Please stay updated by visiting our website: [ICCV23 Challenge at SuperOdometry.com](https://superodometry.com/iccv23_challenge_LiI).

[image]

LarryDong, 2024-03-05:
> Hi @diamonazreal,
>
> The focus of this challenge does not focus on multi-robot systems. Nevertheless, we released small collection of multi-robot data. Specifically, the datasets titled [...]

Hi,

We're also interested in the platform you're using.  
The sequences provided only tells us the "Robot" is UGV1, 2 or 3, but is there reference that show what are these UGVs?

Also, in ICCV 2023 Challenge, we find platform named "RC7" and "SP1", in following links:
[https://superodometry.com/iccv23_challenge_VI](https://superodometry.com/iccv23_challenge_VI)

We've found the hardware information [here](https://github.com/yuanjun-gao/ICCV2023_SLAM_Challenge/blob/master/Hardware_Information.md): 
[image]

I guess "sp1" means spotmini, the 2nd robot. Which one is "RC7"? And are UGV1~3 in this figure?
Hope you could tell us more, thanks.

shibowing, 2024-03-05:
Hi @LarryDong , 

Thanks for the questions.  The RC7 is 
[image]
UGV1~3 are very similar, which can be found here.
[image]

Feel free to let me know if you have more questions.

LarryDong, 2024-03-05:
@shibowing Thanks for your quick reply.

closed 2024-10-29 by shibowing