yuanjun-gao/ICCV2023_SLAM_Challenge #30: Request for Access to the SubT-MRS Dataset Download Link

Gatsby23, 2024-01-28:
Dear Processor,

I recently have read your paper "SubT-MRS Dataset: Pushing SLAM Towards All-weather Environments" and I believe it's a wonderful job and could significantly contribute to the research in the SLAM field. However, I have met some difficulty in downloading this dataset since I could not find any download links available.

I would like to inquire if there are plans to make this dataset publicly accessible in the near future, enabling researchers and developers to download and utilize it for further studies and experimentation. If the dataset is already available for public use, could you please provide the download link?

Thank you for your work and contributions. I look forward to your response.

Sincerely,
Qi Wu.

shibowing, 2024-02-06:
Hi Qi, 
Thank you for your question and interest. The links are available in the Google Drive. Could you please directly send me a link that is not available on your side? I can double-check the link for you. ^ ^ 

Best,

Jonathan-Steinberg, 2024-03-11:
Hi!
As Qi mentioned, the work described at the papaer is indeed wonderful and can help a lot to the field of SLAM.

Though, I can't find the google drive link you have mentioned above..
Can you please add the link for the dataset?

Many thanks for your job, and hopefully I would get the chance to use it as well.

Jonathan

shibowing, 2024-03-11:
Hi  Jonanthan,

Thanks for your interest! Did you check the links form this website?
ICCV 2023 SLAM Challenge - AirLab (superodometry.com)
<https://superodometry.com/datasets>

Best,

Jonathan-Steinberg, 2024-03-11:
Thanks for the quick response!

I have indeed checked the links on the website you have mentioned.

Correct me if I'm wrong, but there is a Google Drive link for every challenge (sensor fusion, virtual inertial & LiDAR).

For example, in [Visual-Inertial challenge](https://superodometry.com/iccv23_challenge_VI), there is a [link ](https://drive.google.com/drive/folders/1drrdOTp7L02QhPJtIvpKo6zB21NoNlYb) for 3 different ROS bags.

My question is, whether there is one folder/link which contains all of the datasets mentioned in the paper, which I have missed, or if the dataset is simply split into the 3 challenges.

Jonathan

shibowing, 2024-10-29:
Hi all, 

We recently cleaned more dataset and released ground truth map. Feel free to check this website 
https://superodometry.com/superloc

Best, 
Shibo

closed 2024-10-29 by shibowing