yuanjun-gao/ICCV2023_SLAM_Challenge #31: groundtruth

liliang2655, 2024-04-02:
[image]
Hello, where can I find the groundtruth files of these data sets? Looking forward to your reply, thank you!

shibowing, 2024-04-17:
Hi all,
Thanks for try our dataset!

The first ICCV SLAM Challenge has concluded. We have summarized the winning teams’ solutions and provided a detailed analysis of the submitted results. For more insights, please review our [Paper](https://arxiv.org/abs/2307.07607).In response to your requests, we have open-sourced our [Ground Truth ](https://drive.google.com/drive/folders/1el51myR4Zzmxz6Tq2lq660zUVQSH9PoQ?usp=sharing)on our website  to facilitate easier evaluation. 

Looking ahead, we will release more datasets with Ground Truth. Stay tuned for more details! To stay informed with the latest updates, be sure to sign up for our mailing [list](https://docs.google.com/forms/d/e/1FAIpQLSfG3YfsC-_qOIATpRJi5qLHDt27j_XyGg-r6fFCn6QIC9kTvg/viewform?vc=0&c=0&w=1&flr=0)

Let me know if there is something missing! 

Best, 

Shibo

liliang2655, 2024-04-20:
> Hi all, Thanks for try our dataset!
>
> The first ICCV SLAM Challenge has concluded. We have summarized the winning teams’ solutions and provided a detailed analysis of the submitted results. For more [...]

Thank you for your efforts. I have another question: Have you calibrated the intrinsic parameters (including noise and drift characteristics) of the IMU within this dataset? In SLAM systems like orb3 and VINS, these IMU intrinsic parameters are essential, but at present, I am unsure about where to find these values.

liliang2655, 2024-04-20:
> Hi all, Thanks for try our dataset!
>
> The first ICCV SLAM Challenge has concluded. We have summarized the winning teams’ solutions and provided a detailed analysis of the submitted results. For more [...]

Could you please tell me where I can find some classic YAML configuration files for SLAM, such as orb3, vinsmono, and so on

xuqingyuan2000, 2024-08-15:
> Could you please tell me where I can find some classic YAML configuration files for SLAM, such as orb3, vinsmono, and so on

I have the same question. Did you solve it?

shibowing, 2024-10-29:
Hi all, sorry for the late reply. I think probalby you can refer below yaml file to tune VINS-MONO

shibowing, 2024-10-29:
`%YAML:1.0

#common parameters
imu_topic: "/imu/data"
image_topic: "/camera_1/image_raw"

depth_topic: "/velodyne_points"
laser_topic: "/velodyne_points"
odom_topic: "/integrated_to_init"
output_path: "/home/shiboz/visual_odometry/VINS-RGBD"

#set parent frame in the config file
lio_world_frame: "sensor_init"
vins_world_frame: "vins_world"
camera_frame: "vins_camera"
lio_laser_frame: "sensor"
laser_frame: "sensor"
vins_imu_frame: "vins_imu"

use_lidar_odom_meas: 1 # 1   means use lidar odometry constraints in vins system
scale_lidar_absolute_factor: 3

align_camera_lidar_estimation: 1 # align camera and lidar estimation for visualization

# undistorted point cloud parameter
odom_frame_id: "world"
laser_frame_id: "lidar"
undistorted_laser_topic: "/points_correct"
voxelSize: 0.1
blindFront: 0.1
blindBack: -1.0
blindLeft: 0.1
blindRight: -0.1

#pointcloud settings
laser_type: 16
pcl_dist: 10
u_boundary: 10
d_boundary: 10
l_boundary: 40
r_boundary: 40
pcl_min_dist: 0.3
pcl_max_dist: 6
resolution: 0.02

model_type: MEI
camera_name: camera
image_width: 640
image_height: 480
mirror_parameters:
   xi: 1.5967783801524584
distortion_parameters:
   k1: -0.04701957438555236
   k2:  0.28058825271927323
   p1:  0.0004063627333225579
   p2:  -0.00015239154341021015
projection_parameters:
   gamma1: 771.6611713881042
   gamma2: 688.7864250675316
   u0: 317.73333375994946
   v0: 233.53737010585292
lidar_to_cam_tx: 0.0
lidar_to_cam_ty: 0.0
lidar_to_cam_tz: 0.0
lidar_to_cam_rx: 0.0
lidar_to_cam_ry: 0.0
lidar_to_cam_rz: 0.0  
# Extrinsic parameter between IMU and Camera.
estimate_extrinsic: 0   # 0  Have an accurate extrinsic parameters. We will trust the following imu^R_cam, imu^T_cam, don't change it.
  # 1  Have an initial guess about extrinsic parameters. We will optimize around your initial guess.
[... 68 lines cut ...]
#for tracked applications
#acc_n: 0.5          # accelerometer measurement noise standard deviation. #0.2
#gyr_n: 0.01         # gyroscope measurement noise standard deviation.     #0.05
#acc_w: 0.001         # accelerometer bias random work noise standard deviation.  #0.02
#gyr_w: 2.0e-5       # gyroscope bias random work noise standard deviation.     #4.0e-5

g_norm: 9.81007       # gravity magnitude

#loop closure parameters
loop_closure: 0                    # start loop closure
fast_relocalization: 0             # useful in real-time and large project
load_previous_pose_graph: 0        # load and reuse previous pose graph; load from 'pose_graph_save_path'
pose_graph_save_path: "/home/shiboz/visual_odometry/VINS-RGBD" # save and load path

#unsynchronization parameters
estimate_td: 1                      # online estimate time offset between camera and imu
td:  -6.905602612688245e-05         # initial value of time offset. unit: s. readed image clock + td = real image clock (IMU clock)
#rolling shutter parameters
rolling_shutter: 0                      # 0: global shutter camera, 1: rolling shutter camera
rolling_shutter_tr: 0.033               # unit: s. rolling shutter read out time per frame (from data sheet).

#visualization parameters
save_image: 0                   # save image in pose graph for visualization prupose; you can close this function by setting 0
visualize_imu_forward: 0        # output imu forward propogation to achieve low latency and high frequence results
visualize_camera_size: 0.4      # size of camera marker in RVIZ`

shibowing, 2024-10-29:
https://drive.google.com/file/d/1GoiBcekTPUaSDPjhs4kNDNgKYGtMrTSV/view?usp=sharing