yuanjun-gao/ICCV2023_SLAM_Challenge #33: Discrepancy Between Ground Truth and IMU Data in Dataset Evaluation

thuThink, 2024-05-28:
# SubT-MRS Dataset
The [dataset](https://drive.google.com/drive/folders/1GR9vavG2iPbXpwlEoH_WzchHspF0ZfQ6) provided for the challenge scenario aligns well with my current research. However, during the process of loading the dataset, I noticed a significant deviation when assessing the pose using IMU data. The dataset's homepage specifies requirements for submitted trajectories, as shown in the image below. I'm curious if the coordinate system of the provided [ground truth results](https://drive.google.com/drive/folders/1qdva6PujKXlLzI5LWDE_kJyS9FYu6JRg?usp=sharing) matches the one depicted in the image.

[image]

If it does, I'm puzzled by the large positional deviation when I use the first position and attitude from the ground truth as the initial values for position and rotation, and then update the position using IMU data's acceleration and angular velocity. If the ground truth data and IMU data share the same coordinate system, it seems apparent that this isn't just an error issue.

[image]
[image]

Below is my code for validating the IMU data. Since it's only a rough simulation without introducing noise and biases, I feel that even so, the simulated IMU data shouldn't deviate so greatly from the ground truth.

```python
import pandas as pd
import yaml
import numpy as np
import matplotlib.pyplot as plt
from scipy.spatial.transform import Rotation as R

# Index(['timestamp', ' q_x', ' q_y', ' q_z', ' q_w', ' ang_vel_x', ' ang_vel_y', ' ang_vel_z', ' lin_acc_x', ' lin_acc_y', ' lin_acc_z'], dtype='object')
# Index(['timestamp', 'p_w_b_x', 'p_w_b_y', 'p_w_b_z', 'q_w_b_x', 'q_w_b_y', 'q_w_b_z', 'q_w_b_w'], dtype='object')

imu_path = './Visual_Inertial_SLAM_Challenge/Handheld1_Folder/imu/imu_data.csv'
gt_path='./Visual_Inertial_SLAM_Challenge/Handheld1_Folder/groundtruth.csv'

imu_data = pd.read_csv(imu_path)
gt_data=pd.read_csv(gt_path)

position = np.array(gt_data[['p_w_b_x', 'p_w_b_y', 'p_w_b_z']].head(1).values)
rotation = R.from_quat(gt_data[['q_w_b_x', 'q_w_b_y', 'q_w_b_z', 'q_w_b_w']].head(1).values).as_matrix()
velocity = (np.array(gt_data[['p_w_b_x', 'p_w_b_y', 'p_w_b_z']].iloc[1].values)-np.array(gt_data[['p_w_b_x', 'p_w_b_y', 'p_w_b_z']].iloc[0].values))/(gt_data['timestamp'].iloc[1]-gt_data['timestamp'].iloc[0])/1e9
DD=5

def compute_position_and_rotation(
    gt_data,
    start_time, 
    end_time, 
    imu_data, 
    position,
    rotation,
    gravity,
    velocity=velocity
    ):
    mask = (imu_data['timestamp'] >= start_time) & (imu_data['timestamp'] <= end_time)
    imu_data = imu_data[mask]
    mask2 = (gt_data['timestamp'] >= start_time) & (gt_data['timestamp'] <= end_time)
    gt_data2 = gt_data[mask2]
    gt_timestamp = gt_data2['timestamp'].values

    closest_indices = [np.abs(imu_data['timestamp'] - t).argmin() for t in gt_timestamp]
    jj=0
    positionx = np.zeros((len(gt_timestamp), 3))
    positionx[jj] = position
    
    for i in range(1, len(imu_data)):
        dt = imu_data['timestamp'].iloc[i] - imu_data['timestamp'].iloc[i - 1]
        dt = dt / 1e9

        angular_velocity = imu_data[[' ang_vel_x', ' ang_vel_y', ' ang_vel_z']].iloc[i-1].values
        acceleration = imu_data[[' lin_acc_x', ' lin_acc_y', ' lin_acc_z']].iloc[i-1].values

        position = position + velocity * dt + 0.5 * (rotation @ (acceleration) - gravity ) * dt**2

        if i in closest_indices:
            jj+=1
            positionx[jj]=position
        # Update velocity and position
        velocity = velocity +  (rotation @ (acceleration) - gravity ) * dt

        # Update rotation
        angular_acceleration= R.from_rotvec((angular_velocity) * dt).as_matrix()
        rotation = rotation @ angular_acceleration

    return positionx, rotation

position, rotation = compute_position_and_rotation(
        gt_data=gt_data,
        start_time=gt_data['timestamp'].iloc[0], 
        end_time=gt_data['timestamp'].iloc[DD+1], 
        imu_data=imu_data, 
        position=position,
        rotation=rotation,
        gravity=np.array([0, 0, -9.81])
    )

```
I've followed the same testing methodology on both the Euroc dataset and the SubT-MRS dataset. Strangely, while the simulated trajectory based on IMU data closely matches the ground truth trajectory in the Euroc dataset, I'm encountering persistent alignment issues in the SubT-MRS dataset.

Considering that the SubT-MRS dataset's ground truth is referenced to the IMU coordinate system, I anticipated a seamless alignment between the IMU data and the ground truth trajectory. It's perplexing that despite this alignment principle, I'm unable to achieve synchronization without discrepancies.

This conundrum has consumed over three weeks of troubleshooting, leaving me at a loss for the root cause of the misalignment. Any insights or suggestions on resolving this issue would be greatly appreciated.
# Tartan Air Datasets
On the homepage of the Tartan Air Datasets, the described ground truth coordinate system is: ["The camera motion is defined in the NED frame. That is to say, the x-axis is pointing to the camera's forward, the y-axis is pointing to the camera's right, the z-axis is pointing to the camera's downward."](https://github.com/castacks/tartanair_tools/blob/master/data_type.md) However, the SubT-MRS dataset describes the ground truth coordinate system as: ["the x-axis is pointing to forward, the y-axis is pointing left, the z-axis is pointing up." ](https://superodometry.com/iccv23_challenge_VI)Since the Tartan Air Datasets are referenced within the SubT-MRS dataset, which coordinate system should be considered authoritative based on the [ground truth](https://drive.google.com/drive/folders/1qdva6PujKXlLzI5LWDE_kJyS9FYu6JRg?usp=sharing) provided on the SubT-MRS dataset homepage?

shibowing, 2024-10-29:
Hi @thuThink ,

Sorry for misleading. I think tartan air dataset has complete different cooridinate system with SubT-MRS. Since Tartan Air dataset is target for camera motion, therefore it is defined in NED frame. However, SubT-MRS uses  the ["the x-axis is pointing to forward, the y-axis is pointing left, the z-axis is pointing up](https://superodometry.com/iccv23_challenge_VI).

shibowing, 2024-10-29:
For trajectory misalignment issue,  for SubT dataset, it should't have such issue.  May I ask which dataset you are suffering this issue? 

Another solution is that  we could use evo to align 2 trajectory togeter. 
https://github.com/MichaelGrupp/evo

closed 2024-10-29 by shibowing